SpikeInterface/spikeinterface #4144: ValueError in get_neuropixels_sample_shifts_from_probe with some probe configs (regression in read_spikeglx)

kendranoneman, 2025-09-16:
I’ve run into a ValueError when using read_spikeglx in recent versions of SpikeInterface. This code used to run without errors in older versions of spike_interface. The issue only occurs for certain probe configurations, not all. 

**Error traceback:**
```
Traceback (most recent call last):
  File "<string>", line 4, in <module>
  File "/ihome/pmayo/knoneman/nhp-data-pipeline/main_pipeline.py", line 26, in run_preprocess
    profile.motion_screening()
  File "/ihome/pmayo/knoneman/nhp-data-pipeline/pipeline/neuropixel.py", line 60, in motion_screening
    raw_recording = read_spikeglx(self.data_path, stream_name=f'imec{self.probe_id}.ap', load_sync_channel=False)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ihome/pmayo/knoneman/envs/NHPipe/lib/python3.11/site-packages/spikeinterface/extractors/neoextractors/spikeglx.py", line 99, in __init__
    sample_shifts = get_neuropixels_sample_shifts_from_probe(probe, stream_name=self.stream_name)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ihome/pmayo/knoneman/envs/NHPipe/lib/python3.11/site-packages/spikeinterface/extractors/neuropixels_utils.py", line 56, in get_neuropixels_sample_shifts_from_probe
    sample_shifts[mux_channels == mux_channel] = np.arange(num_channels_per_adc) / num_cycles_in_adc
    ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: NumPy boolean array indexing assignment cannot assign 12 input values to the 9 output values where the mask is true
```

**Example problematic .ap.meta file:**

acqApLfSy=384,384,1
appVersion=20241215
fileCreateTime=2025-04-28T13:18:58
...
imRowsPerShank=2208
imSampRate=29999.97877094972
...
imroFile=C:/SpikeGLX/Release_v20241215-phase30/SpikeGLX/_Configs/imec_skipRows.imro
nDataDirs=1
nSavedChans=203
snsApLfSy=202,0,1
snsSaveChanSubset=0:157,160:161,164:165,168:169,172:173,176:177,180:181,184:185,188:189,192:193,196:197,200:201,204:205,208:209,212:213,216:217,220:221,224:229,236:237,240:241,244:245,768
syncImInputSlot=2
syncSourceIdx=3
syncSourcePeriod=1
trigMode=Immediate
typeImEnabled=2
typeNiEnabled=1
typeObEnabled=0
typeThis=imec
userNotes=
~imroTbl=(1030,384)(0 0 0 500 250 1)(1 0 0 500 250 1)(2 1 0 500 250 1)(3 1 0 500 250 1)(4 0 0 500 250 1)(5 0 0 500 250 1)(6 1 0 500 250 1)(7 1 0 500 250 1)(8 0 0 500 250 1)(9 0 0 500 250 1)(10 1 0 500 250 1)(11 1 0 500 250 1)(12 0 0 500 250 1)(13 0 0 500 250 1)(14 1 0 500 250 1)(15 1 0 500 250 1)(16 0 0 500 250 1)(17 0 0 500 250 1)(18 1 0 500 250 1)(19 1 0 500 250 1)(20 0 0 500 250 1)(21 0 0 500 250 1)(22 1 0 500 250 1)(23 1 0 500 250 1)(24 0 0 500 250 1)(25 0 0 500 250 1)(26 1 0 500 250 1)(27 1 0 500 250 1)(28 0 0 500 250 1)(29 0 0 500 250 1)(30 1 0 500 250 1)(31 1 0 500 250 1)(32 0 0 500 250 1)(33 0 0 500 250 1)(34 1 0 500 250 1)(35 1 0 500 250 1)(36 0 0 500 250 1)(37 0 0 500 250 1)(38 1 0 500 250 1)(39 1 0 500 250 1)(40 0 0 500 250 1)(41 0 0 500 250 1)(42 1 0 500 250 1)(43 1 0 500 250 1)(44 0 0 500 250 1)(45 0 0 500 250 1)(46 1 0 500 250 1)(47 1 0 500 250 1)(48 0 0 500 250 1)(49 0 0 500 250 1)(50 1 0 500 250 1)(51 1 0 500 250 1)(52 0 0 500 250 1)(53 0 0 500 250 1)(54 1 0 500 250 1)(55 1 0 500 250 1)(56 0 0 500 250 1)(57 0 0 500 250 1)(58 1 0 500 250 1)(59 1 0 500 250 1)(60 0 0 500 250 1)(61 0 0 500 250 1)(62 1 0 500 250 1)(63 1 0 500 250 1)(64 0 0 500 250 1)(65 0 0 500 250 1)(66 1 0 500 250 1)(67 1 0 500 250 1)(68 0 0 500 250 1)(69 0 0 500 250 1)(70 1 0 500 250 1)(71 1 0 500 250 1)(72 0 0 500 250 1)(73 0 0 500 250 1)(74 1 0 500 250 1)(75 1 0 500 250 1)(76 0 0 500 250 1)(77 0 0 500 250 1)(78 1 0 500 250 1)(79 1 0 500 250 1)(80 0 0 500 250 1)(81 0 0 500 250 1)(82 1 0 500 250 1)(83 1 0 500 250 1)(84 0 0 500 250 1)(85 0 0 500 250 1)(86 1 0 500 250 1)(87 1 0 500 250 1)(88 0 0 500 250 1)(89 0 0 500 250 1)(90 1 0 500 250 1)(91 1 0 500 250 1)(92 0 0 500 250 1)(93 0 0 500 250 1)(94 1 0 500 250 1)(95 1 0 500 250 1)(96 0 0 500 250 1)(97 0 0 500 250 1)(98 1 0 500 250 1)(99 1 0 500 250 1)(100 0 0 500 250 1)(101 0 0 500 250 1)(102 1 0 500 250 1)(103 1 0 500 250 1)(104 0 0 500 250 1)(105 0 0 500 250 1)(106 1 0 500 250 1)(107 1 0 500 250 1)(108 0 0 500 250 1)(109 0 0 500 250 1)(110 1 0 500 250 1)(111 1 0 500 250 1)(112 0 0 500 250 1)(113 0 0 500 250 1)(114 1 0 500 250 1)(115 1 0 500 250 1)(116 0 0 500 250 1)(117 0 0 500 250 1)(118 1 0 500 250 1)(119 1 0 500 250 1)(120 0 0 500 250 1)(121 0 0 500 250 1)(122 1 0 500 250 1)(123 1 0 500 250 1)(124 0 0 500 250 1)(125 0 0 500 250 1)(126 1 0 500 250 1)(127 1 0 500 250 1)(128 0 0 500 250 1)(129 0 0 500 250 1)(130 1 0 500 250 1)(131 1 0 500 250 1)(132 0 0 500 250 1)(133 0 0 500 250 1)(134 1 0 500 250 1)(135 1 0 500 250 1)(136 0 0 500 250 1)(137 0 0 500 250 1)(138 1 0 500 250 1)(139 1 0 500 250 1)(140 0 0 500 250 1)(141 0 0 500 250 1)(142 1 0 500 250 1)(143 1 0 500 250 1)(144 0 0 500 250 1)(145 0 0 500 250 1)(146 1 0 500 250 1)(147 1 0 500 250 1)(148 0 0 500 250 1)(149 0 0 500 250 1)(150 1 0 500 250 1)(151 1 0 500 250 1)(152 0 0 500 250 1)(153 0 0 500 250 1)(154 1 0 500 250 1)(155 1 0 500 250 1)(156 0 0 500 250 1)(157 0 0 500 250 1)(158 1 0 500 250 1)(159 1 0 500 250 1)(160 0 0 500 250 1)(161 0 0 500 250 1)(162 1 0 500 250 1)(163 1 0 500 250 1)(164 0 0 500 250 1)(165 0 0 500 250 1)(166 1 0 500 250 1)(167 1 0 500 250 1)(168 0 0 500 250 1)(169 0 0 500 250 1)(170 1 0 500 250 1)(171 1 0 500 250 1)(172 0 0 500 250 1)(173 0 0 500 250 1)(174 1 0 500 250 1)(175 1 0 500 250 1)(176 0 0 500 250 1)(177 0 0 500 250 1)(178 1 0 500 250 1)(179 1 0 500 250 1)(180 0 0 500 250 1)(181 0 0 500 250 1)(182 1 0 500 250 1)(183 1 0 500 250 1)(184 0 0 500 250 1)(185 0 0 500 250 1)(186 1 0 500 250 1)(187 1 0 500 250 1)(188 0 0 500 250 1)(189 0 0 500 250 1)(190 1 0 500 250 1)(191 1 0 500 250 1)(192 0 0 500 250 1)(193 0 0 500 250 1)(194 1 0 500 250 1)(195 1 0 500 250 1)(196 0 0 500 250 1)(197 0 0 500 250 1)(198 1 0 500 250 1)(199 1 0 500 250 1)(200 0 0 500 250 1)(201 0 0 500 250 1)(202 1 0 500 250 1)(203 1 0 500 250 1)(204 0 0 500 250 1)(205 0 0 500 250 1)(206 1 0 500 250 1)(207 1 0 500 250 1)(208 0 0 500 250 1)(209 0 0 500 250 1)(210 1 0 500 250 1)(211 1 0 500 250 1)(212 0 0 500 250 1)(213 0 0 500 250 1)(214 1 0 500 250 1)(215 1 0 500 250 1)(216 0 0 500 250 1)(217 0 0 500 250 1)(218 1 0 500 250 1)(219 1 0 500 250 1)(220 0 0 500 250 1)(221 0 0 500 250 1)(222 1 0 500 250 1)(223 1 0 500 250 1)(224 0 0 500 250 1)(225 0 0 500 250 1)(226 1 0 500 250 1)(227 1 0 500 250 1)(228 0 0 500 250 1)(229 0 0 500 250 1)(230 1 0 500 250 1)(231 1 0 500 250 1)(232 0 0 500 250 1)(233 0 0 500 250 1)(234 1 0 500 250 1)(235 1 0 500 250 1)(236 0 0 500 250 1)(237 0 0 500 250 1)(238 1 0 500 250 1)(239 1 0 500 250 1)(240 0 0 500 250 1)(241 0 0 500 250 1)(242 1 0 500 250 1)(243 1 0 500 250 1)(244 0 0 500 250 1)(245 0 0 500 250 1)(246 1 0 500 250 1)(247 1 0 500 250 1)(248 0 0 500 250 1)(249 0 0 500 250 1)(250 1 0 500 250 1)(251 1 0 500 250 1)(252 0 0 500 250 1)(253 0 0 500 250 1)(254 1 0 500 250 1)(255 1 0 500 250 1)(256 0 0 500 250 1)(257 0 0 500 250 1)(258 1 0 500 250 1)(259 1 0 500 250 1)(260 0 0 500 250 1)(261 0 0 500 250 1)(262 1 0 500 250 1)(263 1 0 500 250 1)(264 0 0 500 250 1)(265 0 0 500 250 1)(266 1 0 500 250 1)(267 1 0 500 250 1)(268 0 0 500 250 1)(269 0 0 500 250 1)(270 1 0 500 250 1)(271 1 0 500 250 1)(272 0 0 500 250 1)(273 0 0 500 250 1)(274 1 0 500 250 1)(275 1 0 500 250 1)(276 0 0 500 250 1)(277 0 0 500 250 1)(278 1 0 500 250 1)(279 1 0 500 250 1)(280 0 0 500 250 1)(281 0 0 500 250 1)(282 1 0 500 250 1)(283 1 0 500 250 1)(284 0 0 500 250 1)(285 0 0 500 250 1)(286 1 0 500 250 1)(287 1 0 500 250 1)(288 0 0 500 250 1)(289 0 0 500 250 1)(290 1 0 500 250 1)(291 1 0 500 250 1)(292 0 0 500 250 1)(293 0 0 500 250 1)(294 1 0 500 250 1)(295 1 0 500 250 1)(296 0 0 500 250 1)(297 0 0 500 250 1)(298 1 0 500 250 1)(299 1 0 500 250 1)(300 0 0 500 250 1)(301 0 0 500 250 1)(302 1 0 500 250 1)(303 1 0 500 250 1)(304 0 0 500 250 1)(305 0 0 500 250 1)(306 1 0 500 250 1)(307 1 0 500 250 1)(308 0 0 500 250 1)(309 0 0 500 250 1)(310 1 0 500 250 1)(311 1 0 500 250 1)(312 0 0 500 250 1)(313 0 0 500 250 1)(314 1 0 500 250 1)(315 1 0 500 250 1)(316 0 0 500 250 1)(317 0 0 500 250 1)(318 1 0 500 250 1)(319 1 0 500 250 1)(320 0 0 500 250 1)(321 0 0 500 250 1)(322 1 0 500 250 1)(323 1 0 500 250 1)(324 0 0 500 250 1)(325 0 0 500 250 1)(326 1 0 500 250 1)(327 1 0 500 250 1)(328 0 0 500 250 1)(329 0 0 500 250 1)(330 1 0 500 250 1)(331 1 0 500 250 1)(332 0 0 500 250 1)(333 0 0 500 250 1)(334 1 0 500 250 1)(335 1 0  [... cut ...] (0:114:1720:1)(0:11:1760:1)(0:114:1760:1)(0:11:1800:1)(0:114:1800:1)(0:11:1840:1)(0:114:1840:1)(0:11:1880:1)(0:114:1880:1)(0:11:1920:1)(0:114:1920:1)(0:11:1960:1)(0:114:1960:1)(0:11:2000:1)(0:114:2000:1)(0:11:2040:1)(0:114:2040:1)(0:11:2080:1)(0:114:2080:1)(0:11:2120:1)(0:114:2120:1)(0:11:2160:1)(0:114:2160:1)(0:11:2200:1)(0:114:2200:1)(0:11:2240:1)(0:114:2240:1)(0:11:6100:1)(0:114:6100:1)(0:11:2280:1)(0:114:2280:1)(0:11:2360:1)(0:114:2360:1)(0:11:2400:1)(0:114:2400:1)(0:11:2440:1)(0:114:2440:1)

chrishalcrow, 2025-09-17:
Hello,

Ask @zm711 said, we've significantly updated this bit of code. Before, spikeinterface made some guesses about your probe to figure out which channels were sampled together (let's call the channels that are sampled together sample-groups). These guesses were correct for the basic NP1 and 2 probes, but we couldn't test edge cases, and if some new probe design came along the guess would probably be wrong. So now we use "mux tables", which tell us the sample-groups, straight from the engineers (https://github.com/billkarsh/probetable/). This new method should be less likely to be incorrect, means more probes are supported, and any new probes are automatically supported.

With that in mind, the old code was making a guess about your probe and assigning sample-groups based on this guess. I'm almost sure this guess was incorrect. The error your seeing now is a hint that we don't know the sample-groups for the "interesting/accidentally didn't record from 385 channels" probe configurations. It would be great to figure out a way to support this (I'll have a think!). In the meantime, your safest bet is to not apply a `phase_shift` for the interesting probes (this is a bit hideous: but you could `try/except` this preprocessing step...)

alejoe91, 2025-09-17:
Can you attach the full meta file?

kendranoneman, 2025-09-18:
GitHub won't allow me to attach a .meta file, but here are the contents:

acqApLfSy=384,384,1                                                                                                                                                                                                                                                              
appVersion=20241215
fileCreateTime=2025-04-28T13:18:58
fileName=D://kendra_scrappy_0139a_g0/kendra_scrappy_0139a_g0_imec0/kendra_scrappy_0139a_g0_t0.imec0.ap.bin
fileSHA1=533C527CDCEDDC321BC9C2CE12E4B51BB1A35D18
fileSizeBytes=86052444198
fileTimeSecs=7065.066099488115
firstSample=145379487
gateMode=Immediate
imAiRangeMax=0.6
imAiRangeMin=-0.6
imAnyChanFullBand=false
imCalibrated=true
imChan0Ref=ext
imChan0apGain=500
imColsPerShank=2
imDatApi=3.70.3
imDatBs_fw=2.0.169
imDatBsc_fw=3.2.191
imDatBsc_hw=2.2
imDatBsc_pn=NP2_QBSC_00
imDatBsc_sn=23131263
imDatFx_hw=3.6
imDatFx_pn=NPNH_AFLEX_00
imDatFx_sn=0
imDatHs_fw=0.0.0
imDatHs_hw=5.3
imDatHs_pn=NP2_HS_30
imDatHs_sn=23060176
imDatPrb_dock=1
imDatPrb_pn=NP1032
imDatPrb_port=1
imDatPrb_slot=2
imDatPrb_sn=23096407202
imDatPrb_type=1030
imErrFlags0_IS_CT_SR_LK_PP_SY=0 0 0 0 0 0
imErrFlags1_IS_CT_SR_LK_PP_SY=0 0 0 0 0 0
imIsSvyRun=false
imLEDEnable=false
imLowLatency=false
imMaxInt=512
imRowsPerShank=2208
imSampRate=29999.97877094972
imStdby=
imSvyMaxBnk=-1
imSvyNShanks=1
imSvySecPerBnk=35
imTipLength=373
imTrgRising=true
imTrgSource=0
imX0EvenRow=11
imX0OddRow=11
imXPitch=103
imZPitch=20
imroFile=C:/SpikeGLX/Release_v20241215-phase30/SpikeGLX/_Configs/imec_skipRows.imro
nDataDirs=1
nSavedChans=203
snsApLfSy=202,0,1
snsSaveChanSubset=0:157,160:161,164:165,168:169,172:173,176:177,180:181,184:185,188:189,192:193,196:197,200:201,204:205,208:209,212:213,216:217,220:221,224:229,236:237,240:241,244:245,768
syncImInputSlot=2
syncSourceIdx=3
syncSourcePeriod=1
trigMode=Immediate
typeImEnabled=2
typeNiEnabled=1
typeObEnabled=0
typeThis=imec
userNotes=
~imroTbl=(1030,384)(0 0 0 500 250 1)(1 0 0 500 250 1)(2 1 0 500 250 1)(3 1 0 500 250 1)(4 0 0 500 250 1)(5 0 0 500 250 1)(6 1 0 500 250 1)(7 1 0 500 250 1)(8 0 0 500 250 1)(9 0 0 500 250 1)(10 1 0 500 250 1)(11 1 0 500 250 1)(12 0 0 500 250 1)(13 0 0 500 250 1)(14 1 0 500 250 1)(15 1 0 500 250 1)(16 0 0 500 250 1)(17 0 0 500 250 1)(18 1 0 500 250 1)(19 1 0 500 250 1)(20 0 0 500 250 1)(21 0 0 500 250 1)(22 1 0 500 250 1)(23 1 0 500 250 1)(24 0 0 500 250 1)(25 0 0 500 250 1)(26 1 0 500 250 1)(27 1 0 500 250 1)(28 0 0 500 250 1)(29 0 0 500 250 1)(30 1 0 500 250 1)(31 1 0 500 250 1)(32 0 0 500 250 1)(33 0 0 500 250 1)(34 1 0 500 250 1)(35 1 0 500 250 1)(36 0 0 500 250 1)(37 0 0 500 250 1)(38 1 0 500 250 1)(39 1 0 500 250 1)(40 0 0 500 250 1)(41 0 0 500 250 1)(42 1 0 500 250 1)(43 1 0 500 250 1)(44 0 0 500 250 1)(45 0 0 500 250 1)(46 1 0 500 250 1)(47 1 0 500 250 1)(48 0 0 500 250 1)(49 0 0 500 250 1)(50 1 0 500 250 1)(51 1 0 500 250 1)(52 0 0 500 250 1)(53 0 0 500 250 1)(54 1 0 500 250 1)(55 1 0 500 250 1)(56 0 0 500 250 1)(57 0 0 500 250 1)(58 1 0 500 250 1)(59 1 0 500 250 1)(60 0 0 500 250 1)(61 0 0 500 250 1)(62 1 0 500 250 1)(63 1 0 500 250 1)(64 0 0 500 250 1)(65 0 0 500 250 1)(66 1 0 500 250 1)(67 1 0 500 250 1)(68 0 0 500 250 1)(69 0 0 500 250 1)(70 1 0 500 250 1)(71 1 0 500 250 1)(72 0 0 500 250 1)(73 0 0 500 250 1)(74 1 0 500 250 1)(75 1 0 500 250 1)(76 0 0 500 250 1)(77 0 0 500 250 1)(78 1 0 500 250 1)(79 1 0 500 250 1)(80 0 0 500 250 1)(81 0 0 500 250 1)(82 1 0 500 250 1)(83 1 0 500 250 1)(84 0 0 500 250 1)(85 0 0 500 250 1)(86 1 0 500 250 1)(87 1 0 500 250 1)(88 0 0 500 250 1)(89 0 0 500 250 1)(90 1 0 500 250 1)(91 1 0 500 250 1)(92 0 0 500 250 1)(93 0 0 500 250 1)(94 1 0 500 250 1)(95 1 0 500 250 1)(96 0 0 500 250 1)(97 0 0 500 250 1)(98 1 0 500 250 1)(99 1 0 500 250 1)(100 0 0 500 250 1)(101 0 0 500 250 1)(102 1 0 500 250 1)(103 1 0 500 250 1)(104 0 0 500 250 1)(105 0 0 500 250 1)(106 1 0 500 250 1)(107 1 0 500 250 1)(108 0 0 500 250 1)(109 0 0 500 250 1)(110 1 0 500 250 1)(111 1 0 500 250 1)(112 0 0 500 250 1)(113 0 0 500 250 1)(114 1 0 500 250 1)(115 1 0 500 250 1)(116 0 0 500 250 1)(117 0 0 500 250 1)(118 1 0 500 250 1)(119 1 0 500 250 1)(120 0 0 500 250 1)(121 0 0 500 250 1)(122 1 0 500 250 1)(123 1 0 500 250 1)(124 0 0 500 250 1)(125 0 0 500 250 1)(126 1 0 500 250 1)(127 1 0 500 250 1)(128 0 0 500 250 1)(129 0 0 500 250 1)(130 1 0 500 250 1)(131 1 0 500 250 1)(132 0 0 500 250 1)(133 0 0 500 250 1)(134 1 0 500 250 1)(135 1 0 500 250 1)(136 0 0 500 250 1)(137 0 0 500 250 1)(138 1 0 500 250 1)(139 1 0 500 250 1)(140 0 0 500 250 1)(141 0 0 500 250 1)(142 1 0 500 250 1)(143 1 0 500 250 1)(144 0 0 500 250 1)(145 0 0 500 250 1)(146 1 0 500 250 1)(147 1 0 500 250 1)(148 0 0 500 250 1)(149 0 0 500 250 1)(150 1 0 500 250 1)(151 1 0 500 250 1)(152 0 0 500 250 1)(153 0 0 500 250 1)(154 1 0 500 250 1)(155 1 0 500 250 1)(156 0 0 500 250 1)(157 0 0 500 250 1)(158 1 0 500 250 1)(159 1 0 500 250 1)(160 0 0 500 250 1)(161 0 0 500 250 1)(162 1 0 500 250 1)(163 1 0 500 250 1)(164 0 0 500 250 1)(165 0 0 500 250 1)(166 1 0 500 250 1)(167 1 0 500 250 1)(168 0 0 500 250 1)(169 0 0 500 250 1)(170 1 0 500 250 1)(171 1 0 500 250 1)(172 0 0 500 250 1)(173 0 0 500 250 1)(174 1 0 500 250 1)(175 1 0 500 250 1)(176 0 0 500 250 1)(177 0 0 500 250 1)(178 1 0 500 250 1)(179 1 0 500 250 1)(180 0 0 500 250 1)(181 0 0 500 250 1)(182 1 0 500 250 1)(183 1 0 500 250 1)(184 0 0 500 250 1)(185 0 0 500 250 1)(186 1 0 500 250 1)(187 1 0 500 250 1)(188 0 0 500 250 1)(189 0 0 500 250 1)(190 1 0 500 250 1)(191 [... cut ...] (0:114:1720:1)(0:11:1760:1)(0:114:1760:1)(0:11:1800:1)(0:114:1800:1)(0:11:1840:1)(0:114:1840:1)(0:11:1880:1)(0:114:1880:1)(0:11:1920:1)(0:114:1920:1)(0:11:1960:1)(0:114:1960:1)(0:11:2000:1)(0:114:2000:1)(0:11:2040:1)(0:114:2040:1)(0:11:2080:1)(0:114:2080:1)(0:11:2120:1)(0:114:2120:1)(0:11:2160:1)(0:114:2160:1)(0:11:2200:1)(0:114:2200:1)(0:11:2240:1)(0:114:2240:1)(0:11:6100:1)(0:114:6100:1)(0:11:2280:1)(0:114:2280:1)(0:11:2360:1)(0:114:2360:1)(0:11:2400:1)(0:114:2400:1)(0:11:2440:1)(0:114:2440:1)

closed 2025-10-24 by chrishalcrow