SpikeInterface/spikeinterface #4150: Instability with SC2 in dense recordings

b-grimaud, 2025-09-26:
Sorry to bring this up again, as I'm pretty sure this is only an issue on recordings with a few thousands of units.

I did not keep up with new commits for a while, and when trying to run sorting again on the latest version it systematically crashes.

The weird thing is that the point at which it crashes is not always reproducible. Starting the script from a local session, over SSH or in a tmux server also seems to have an influence.

The most common crash points are : 
 - Creating the `mask` array in `compute_similarity_with_templates_array` ([here](https://github.com/SpikeInterface/spikeinterface/blob/2e1f280ff82bf737ce5fcd512c74ba425043962a/src/spikeinterface/postprocessing/template_similarity.py#L386))
 - During the main loop of `_compute_similarity_matrix_numba` ([here](https://github.com/SpikeInterface/spikeinterface/blob/2e1f280ff82bf737ce5fcd512c74ba425043962a/src/spikeinterface/postprocessing/template_similarity.py#L301))
 
Looking at the resource usage, the pattern is always the same : there are several Python processes, the number of which is equal to `n_jobs`, with an equal amount of memory usage, and another Python process with a lot more memory (20-30GB). Then, that amount rapidly climbs to to ~50-60GB, then the session crashes. I'm not entirely sure this is an OOM error, because this amount is still small enough to fit in memory. So either it rises so fast that the session crashes before the display has time to update, or something else is happening.

As a side note, `_get_optimal_n_jobs` works properly and `n_jobs` is scaled to memory appropriately. I've noticed that it is only used when `templates_from_svd` is set to `False` but I'm not quite sure of the difference.

b-grimaud, 2025-09-29:
Update : I also get a similar behavior with TdC2 at this step 

> Creating the mask array in compute_similarity_with_templates_array ([here](https://github.com/SpikeInterface/spikeinterface/blob/2e1f280ff82bf737ce5fcd512c74ba425043962a/src/spikeinterface/postprocessing/template_similarity.py#L386))

So this doesn't seem to be sorter-specific. Oddly, it only seems to happen if I use more than one core and/or start the script from a tmux  session. If I start it directly from the shell with `n_jobs` set to 1 the sorting goes on, albeit slowly.

b-grimaud, 2025-09-29:
That's about it, 3000 to 5000 units on average for _in vitro_ recordings. I can share recordings if needed, but at around 8 GB/minute they're quite heavy. I have 1, 5 and 10 minutes recordings. I also have the probe geometry exported as YAML and JSON if needed.

yger, 2025-09-29:
I don't think you need to share anything. Based on your comments, there seems to be an issue in compute_similarity_with_templates_array because of internal data structure that should be too large. I'll dig more in depth

b-grimaud, 2025-10-02:
Tested it and it does fix the problem, thanks !

> I think you should turn this option to True both for speed and memory considerations

Good to know, I'll stick with it going forward.

There's another, similar problem happening down the line.
I've narrowed it down to the initialization of `CircusOMPSVDPeeler` in `_prepare_templates`, specifically when calculating `unit_overlaps` [here](https://github.com/SpikeInterface/spikeinterface/blob/2e1f280ff82bf737ce5fcd512c74ba425043962a/src/spikeinterface/sortingcomponents/matching/circus.py#L247).

I can open a separate issue if needed.

yger, 2025-10-03:
The memory footprint for overlaps is also fixed in #4157

b-grimaud, 2025-10-06:
I haven't had time to pinpoint the issue exactly, but I think it might be the sparsity estimation when the analyzer is created.

If I create an analyzer from scratch on this kind of high-density sorting with `n_jobs` to the max, I get a crash.

If I do the same with `n_jobs=1` the analyzer is created just fine, and I can switch back to parallel processing to compute the extensions.

If I reload the analyzer created by `final_cleaning_circus` it also works, but maybe sparsity is also reloaded in that case ?