SpikeInterface/spikeinterface #4153: HerdingSpikes Spike Sorter runtime error using the latest docker image

jorgeclarocardoso, 2025-09-30:
Hi all, 

I'm trying to run the HerdingSpikes spike sorter from the latest docker image available and I'm getting this error. What's the best approach to fix this issue and get it running?

Thank you!

installation_mode='auto' switching to installation_mode: 'github'
Starting container
Installing spikeinterface with github in container
Running herdingspikes sorter inside spikeinterface/herdingspikes-base
Stopping container
Traceback (most recent call last):
  File "/home/user/code/python/spikeinterface/howto/./check_docker_installation.py", line 34, in <module>
    my_pipeline()
  File "/home/user/code/python/spikeinterface/howto/./check_docker_installation.py", line 20, in my_pipeline
    sorting = si.run_sorter(sorter_name='herdingspikes',
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/anaconda3/envs/si_env/lib/python3.12/site-packages/spikeinterface/sorters/runsorter.py", line 204, in run_sorter
    return run_sorter_container(
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/anaconda3/envs/si_env/lib/python3.12/site-packages/spikeinterface/sorters/runsorter.py", line 681, in run_sorter_container
    raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/home/user/code/python/spikeinterface/howto/in_container_sorter_script.py", line 23, in <module>
    sorting = run_sorter_local(
              ^^^^^^^^^^^^^^^^^
  File "/root/.local/lib/python3.12/site-packages/spikeinterface/sorters/runsorter.py", line 302, in run_sorter_local
    SorterClass.run_from_folder(folder, raise_error, verbose)
  File "/root/.local/lib/python3.12/site-packages/spikeinterface/sorters/basesorter.py", line 263, in run_from_folder
    "sorter_version": str(SorterClass.get_sorter_version()),
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.local/lib/python3.12/site-packages/spikeinterface/sorters/external/herdingspikes.py", line 109, in get_sorter_version
    import herdingspikes as hs
  File "/usr/local/lib/python3.12/site-packages/herdingspikes/__init__.py", line 4, in <module>
    from herdingspikes.hs2 import HSDetection, HSClustering, HSDetectionLightning
  File "/usr/local/lib/python3.12/site-packages/herdingspikes/hs2.py", line 5, in <module>
    from sympy import N
ModuleNotFoundError: No module named 'sympy'

zm711, 2025-10-04:
@alejoe91 I always forget how to install extra packages for docker. Could you take this one? @jorgeclarocardoso we do have a mechanism for this but I forget sorry!

alejoe91, 2025-10-06:
> [@alejoe91](https://github.com/alejoe91) I always forget how to install extra packages for docker. Could you take this one? [@jorgeclarocardoso](https://github.com/jorgeclarocardoso) we do have a mechanism for this but I forget sorry!

The mechamism is `recording.extra_requirements.append("missing-package")`. But this seems an issue with the docker image! I'll check!

alejoe91, 2025-10-06:
Pushed a new docker version. Can you try to pull again latest or specifically use `docker_image="spikeinterface/herdingspikes-base:0.4.6"? @jorgeclarocardoso 

I tested locally and it works so I'll close the issue. Feel free to reopen!